circusautomatic/cthuloid #5: Hardware issue when trying to control an rgb led with the tlc59711

plankatron, 2017-10-28:
From various tests I believe there is a problem in how the light board pcb is designed.
It initially came up when controlling a single 100w led at 10khz with a 16 bit pwm resolution. We found that the useable values were constrained by about half, this is to say that visually the top or low end values would yield no real visual difference in light intensity. We fixed this in a way by running at 1khz  and a 470uf cap on the output for  smoothing which resulted in a much broader range of useable values.

Now we are using the tlc59711 led driver to generate a high frequency and high bit signal and the output is severely restricted, initial thoughts of the hardware problem being the Arduino start to seem false and that the real culprit is the lightboard design. Thus this issue is being started to try to figure out the problem and to design a solution.

plankatron, 2017-10-28:
# Basic test
* Make sure you are running the latest Arduino ide (older versions could work but have not been tested) and have loaded this library for the tlc59711 https://github.com/ulrichstern/Tlc59711/
* Have a 100w led being controlled by an Arduino uno or mega that is using the tlc59711 as a driver
*  Load the helloworld example sketch but make sure in void loop that it is just setting a single value like 
**  tlc.setRGB(65535, 65535,65535);
**  tlc.write();
* Keep changing values to see that the only really useful pwm values are in a tiny range.

Now the real work of figuring out what is causing this and how to get a much larger range from this signal.

plankatron, 2017-10-28:
Tlc59711 info and data sheets : http://www.ti.com/product/TLC59711
Adafruit product page : https://www.adafruit.com/product/1455
Wiring to Arduino : https://learn.adafruit.com/tlc5947-tlc59711-pwm-led-driver-breakout/connecting-to-the-arduino

flintiii, 2017-10-28:
So would it be useful to eliminate the Arduino?

plankatron, 2017-10-28:
[image]
Also a link to the parts list :
http://organize.7rnx.net/projects/sad-robot-workspace/wiki/PCB#Parts-List

plankatron, 2017-10-28:
Thinking that this could work as a simplified circuit design but it still needs to invert the signal somehow?

[image]

flintiii, 2017-10-29:
These pictures are excellent!

flintiii, 2017-10-29:
I am wondering if you can accomplish the inversion you need in software...

plankatron, 2017-11-15:
info for dimming led's with the esp32 
https://techtutorialsx.com/2017/06/15/esp32-arduino-led-pwm-fading/

closed 2018-01-19 by plankatron